christianrauch/apriltag_ros #1: Feature request: Adding /tag_detections_image topic

kgallowa, 2020-08-04:
Thanks for the nice work on this package! I was able to get everything working, but I noticed that the running node doesn't publish to a /tag_detections_image topic like the ROS1 version (from AprilRobotics/apriltag_ros) does for helpful visualization. Any plans to add that?

christianrauch, 2020-08-04:
The `apriltag_ros` node only processes AprilTags and publishes the tag poses and raw AprilTag detections. There is a dedicated [`apriltag_viz` node](https://github.com/christianrauch/apriltag_viz) that uses the raw detections and publishes a visualisation on topic `tag_detections_image`.

This is intentional to keep the core processing node small. Since both noes implement the `rclcpp::Node` interface, you can put them into the same process with minimal / no IPC overhead. You could write a custom launch script that behaves like the ROS1 node and combines both as `ComposableNode` in a single `ComposableNodeContainer`.

kgallowa, 2020-08-04:
Great; thanks! I hadn't seen the apriltag_viz package. Makes sense to keep those functions separate.

closed 2020-08-04 by kgallowa