christianrauch/apriltag_ros #2: building on Foxy Fitzroy, Ubuntu 20.04

b0hne, 2020-08-12:
In order to build the node on Foxy i had to change CMakeLists.txt, line 26:

-target_link_libraries(AprilTagNode apriltag::apriltag)
+target_link_libraries(AprilTagNode apriltag)

works fine afterwards. Thank you for your work

christianrauch, 2020-08-12:
`apriltag` on `master` uses the `apriltag::` namespace since commit https://github.com/AprilRobotics/apriltag/commit/35f4ef2825745a08ec4690ce8c15f8dc903c2868.

If you want to use the release `apriltag` version 3.1.2, you have to use the stable `apriltag_ros` version 2.1.0 (or `master` up to d8d533406995c46d2de30de6401d61c052c6e3ca). Otherwise, you have to use the current `master` of both packages.

christianrauch, 2021-04-29:
I assume this issue has been resolved when using the `master` branch. I am going to close this.

If the problem re-appears, re-open this issue or provide a PR. Thanks.

closed 2021-04-29 by christianrauch

b0hne, 2021-07-24:
Sorry, i haven't revisited the issue for a while. I had installed ros-foxy-apriltag via apt before and that messed up the target_link_libraries command. Thank you for the answer.