christianrauch/apriltag_ros #3: Failed to load library libapriltag.so.3

liushuya7, 2021-04-29:
Hi,

Everything compiled fine with colcon build. However, when I tried to ```ros2 launch apriltag_ros tag_16h5_all.launch.py```:
```
[INFO] [component_container-1]: process started with pid [180093]
[component_container-1] [INFO] [1619718701.105940189] [apriltag.tag_container]: Load Library: /home/josh/dev_ws/install/apriltag_ros/lib/libAprilTagNode.so
[component_container-1] [ERROR] [1619718701.109247249] [apriltag.tag_container]: Failed to load library: Could not load library LoadLibrary error: libapriltag.so.3: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-foxy-rcutils-1.1.3/src/shared_library.c:84
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'apriltag' of type 'AprilTagNode' in container '/apriltag/tag_container': Failed to load library: Could not load library LoadLibrary error: libapriltag.so.3: cannot open shared object file: No such file or directory, at /tmp/binarydeb/ros-foxy-rcutils-1.1.3/src/shared_library.c:84
```
I have installed Apritag to the system, and I have /usr/local/lib/libapriltag.so.3
Would you know where I may have done wrong?
Thanks,
Joshua

liushuya7, 2021-04-29:
Running ```sudo ldconfig``` resolved it.

closed 2021-04-29 by liushuya7

christianrauch, 2021-04-29:
It appears you installed the `apriltag` library from https://github.com/AprilRobotics/apriltag directly as a system package via CMake. I think you can avoid this issue by installing it as part of your colon workspace.

The easiest would be to just to install the binary package `ros-foxy-apriltag` :-)