christianrauch/apriltag_ros #9: Jetson OpenCV issue

trentatsa, 2022-07-21:
Hi There, I am currently going around in circles. I am trying to run your package on Jetson Xavier NX. The package launches but as soon its it receives and image the following error is thrown:

[component_container-1] terminate called after throwing an instance of 'cv::Exception'
[component_container-1]   what():  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/matrix.cpp:250: error: (-215:Assertion failed) s >= 0 in function 'setSize'
[component_container-1] 
[ERROR] [component_container-1]: process has died [pid 37461, exit code -6, cmd '/opt/ros/foxy/lib/rclcpp_components/component_container --ros-args -r __node:=tag_container -r __ns:=/apriltag'].

I think it is something do with the OpenCV version, where the Jetson Xavier has 4.5 inbuild. I have tried to install Jetson OS without OpenCV and install OpenCV with (sudo apt install libopencv-dev python3-opencv).

christianrauch, 2022-07-21:
I haven't encountered this error nor do I have the hardware to test this. Could you provide a full backtrace of the error?

Btw, it makes it easier to read if you format code and terminal output as "code" (via backticks).

christianrauch, 2022-08-27:
There is not enough information to debug this issue. It is unclear if this is an issue with the apriltag library or this node.

I suggest that you [install the `dbgsym` packages](https://wiki.ubuntu.com/Debug%20Symbol%20Packages) for OpenCV and any other libraries in the stack.

I am closing this for now. If this occurs again, please reopen the issue and provide a detailed backtrace and maybe a reproducible example with a bag file.

closed 2022-08-27 by christianrauch

orjano-max, 2023-02-14:
**Posting here as i had a similar issue and my solution might help others that are stuck with this.**

**Using** 
- Nvidia AGX Xavier 
- Jetpack 5.0.2
-  ROS2 Foxy.

Encountered the issue described in #2, did the fix, but got complaints about missing apriltag_msgs.
Decided to use the following git repo which is inspired by this one(but has a foxy branch and contains apriltag_msgs package): https://github.com/Adlink-ROS/apriltag_ros

**Successfully built packages and got the issue described by @trentatsa here.**

Solved this issue by cloning and building the **vision_opencv** packages from source and using these instead of binaries:

https://github.com/ros-perception/vision_opencv

Of course using the foxy branch as it corresponds to the ROS2 distro i am using.

christianrauch, 2023-02-14:
> Encountered the issue described in #2, did the fix, but got complaints about missing apriltag_msgs.

FYI, it is here: https://github.com/christianrauch/apriltag_msgs. I should probably document the installation from source.

> Solved this issue by cloning and building the **vision_opencv** packages from source and using these instead of binaries:

Thanks for the tip. I assume that other packages using `vision_opencv` (or `cv_bridge` specifically) would also exhibit this issue.

Can it be that there are multiple versions of OpenCV installed on your system? If you have multiple versions of OpenCV installed and `libcv_bridge.so` selects the wrong one, then this is likely causing this issue. Make sure to only have the OpenCV version installed that is also used by the `ros-$ROS_DISTRO-cv-bridge` package.

orjano-max, 2023-02-15:
@christianrauch Yeah, Jetpack 5.0.2 comes with opencv 4.5.4 preinstalled, ref: https://developer.nvidia.com/embedded/vpi
Which might cause problems.