christianrauch/apriltag_ros #10: ros2 echo, message is invalid

lorepieri8, 2022-08-10:
```
$ ros2 topic list
/apriltag/detections
/clicked_point
/color/camera_info
/color/image_raw
/color/metadata
/depth/camera_info
/depth/image_rect_raw
/depth/metadata
/extrinsics/depth_to_color
/extrinsics/depth_to_infra1
/extrinsics/depth_to_infra2
/goal_pose
/imu
/infra1/camera_info
/infra1/image_rect_raw
/infra1/metadata
/infra2/camera_info
/infra2/image_rect_raw
/infra2/metadata
/initialpose
/parameter_events
/rosout
/tf
/tf_static
```

```
$ ros2 topic echo /apriltag/detections 
The message type 'apriltag_msgs/msg/AprilTagDetectionArray' is invalid

```

christianrauch, 2022-08-10:
Did you source the workspace that contains that message definition?

lorepieri8, 2022-08-12:
That was the problem, thanks.

closed 2022-08-12 by lorepieri8