christianrauch/apriltag_ros #16: OutOfMemoryError with pose branch

TomGrimwood, 2023-06-25:
Using the pose branch, when the image stream begins, I receive this error;

```html
[INFO] [launch]: All log files can be found below /home/user/.ros/log/2023-06-26-11-15-48-213828-node1-orin-2340600
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container_mt-1]: process started with pid [2340628]
[component_container_mt-1] [INFO] [1687734948.719613070] [apriltag_container_detect]: Load Library: /home/user/workspaces/april_ws/install/apriltag_ros/lib/libAprilTagNode.so
[component_container_mt-1] [INFO] [1687734948.831306213] [apriltag_container_detect]: Found class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[component_container_mt-1] [INFO] [1687734948.831585384] [apriltag_container_detect]: Instantiate class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/apriltag' in container '/apriltag_container_detect'
[component_container_mt-1] terminate called after throwing an instance of 'cv::Exception'
[component_container_mt-1]   what():  OpenCV(4.5.4) /home/ubuntu/build_opencv/opencv/modules/core/src/alloc.cpp:73: error: (-4:Insufficient memory) Failed to allocate 117094104464640 bytes in function 'OutOfMemoryError'
[component_container_mt-1] 
[ERROR] [component_container_mt-1]: process has died [pid 2340628, exit code -6, cmd '/opt/ros/galactic/lib/rclcpp_components/component_container_mt --ros-args -r __node:=apriltag_container_detect -r __ns:=/'].
```

I am using an Nvidia Orin with OpenCV 4.5.4, ROS Galactic.
If you need any further information let me know.

christianrauch, 2023-06-26:
Well, I guess your Jetson Orin does not have 117094104464640 bytes of memory :-)

It's unclear to me why this happens. I assume that at some point the image dimensions are wrongly calculated or there is an integer overflow.

Do you see this issue only on the `pose` branch? Can you provide a rosbag to reproduce this and/or a gdb backtrace?

christianrauch, 2023-11-19:
@TomGrimwood Can you clarify if this only happens on the `pose` branch? Can you provide a log file to reproduce this? Alternatively, can you provide a backtrace with source code line numbers and the commit at which this happens?

sebastianserna30, 2025-01-29:
Just for completeness:

I also had this error. What helped me was, like mentioned in [here](https://github.com/christianrauch/apriltag_ros/issues/9#issuecomment-1429778790),  building [vision_opencv](https://github.com/ros-perception/vision_opencv) from source.