christianrauch/apriltag_ros #17: Apriltag ROS2 package can not be found

wangzizhe, 2023-07-07:
Hi Christian,

I found Apriltag for ROS2 could be installed using `sudo apt-get install ros-humble-apriltag` (my ROS2 version is humble), it will display that the installation is successful, but you can’t find this package using `ros2 pkg list`, under `opt/ros/humble/share` you can see the apriltag folder, but when you go into it, you find it is actually not a package, inside there is only a cmake folder.

It is somehow strange, or I did something wrong?

Thanks in advance!

christianrauch, 2023-07-10:
`ros-humble-apriltag` is just the apriltag library from https://github.com/AprilRobotics/apriltag, which is a regular CMake package with a `package.xml`. Only ament packages will be registered and listed as a ROS packages. The ROS node can be installed with package `ros-humble-apriltag-msgs`.

wangzizhe, 2023-07-14:
Thank you for your answer Christian.

As I understood right, use s`udo apt-get install ros-humble-apriltag-msgs `or manually clone this repo and build it.
They are actually the same thing. 

Is it right?

wangzizhe, 2023-07-14:
Thx

Now I figured out the differences:

* `ros-humble-apriltag`
    * https://github.com/AprilRobotics/apriltag
    * AprilTag is a visual fiducial system popular in robotics research. This repository contains the most recent version of AprilTag, AprilTag 3, which includes a faster (>2x) detector, improved detection rate on small tags, flexible tag layouts, and pose estimation. AprilTag consists of a small C library with minimal dependencies

* https://github.com/AprilRobotics/apriltag_ros
    * A ROS wrapper of the AprilTag 3 visual fiducial detector 

* https://github.com/christianrauch/apriltag_ros
    * ROS2 node for AprilTag detection 

* `ros-humble-apriltag-msgs`
    * AprilTag message definitions

closed 2023-07-14 by wangzizhe