christianrauch/apriltag_ros #18: v4l2_36h11.launch.yml | Error “Rectified topic ‘/v4l2/image_rect’ requested but camera publishing ‘/v4l2/camera_info’ is uncalibrated”

wangzizhe, 2023-09-01:
Hi Christian,

### Background

Camera was calibrated and run 

`ros2 run v4l2_camera v4l2_camera_node`

is no problem.

---

### Issue

Now I'm trying to run the apriltag node with the example launch file:

`ros2 launch apriltag_ros v4l2_36h11.launch.yml`

The command uses `.yml` to configure, this `.yml` concludes:

```
launch:
- arg:
    name: device
    default: "0"

- node_container:
    pkg: rclcpp_components
    exec: component_container
    name: apriltag_container
    namespace: ""
    composable_node:
    - pkg: v4l2_camera
      plugin: v4l2_camera::V4L2Camera
      name: camera
      namespace: v4l2
      param:
      - name: video_device
        value: /dev/video$(var device)
      extra_arg:
      - name: use_intra_process_comms
        value: "True"

    - pkg: image_proc
      plugin: image_proc::RectifyNode
      name: rectify
      namespace: v4l2
      remap:
      - from: image
        to: image_raw
      extra_arg:
      - name: use_intra_process_comms
        value: "True"

    - pkg: apriltag_ros
      plugin: AprilTagNode
      name: apriltag
      namespace: apriltag
      remap:
      - from: /apriltag/image_rect
        to: /v4l2/image_rect
      - from: /apriltag/camera_info
        to: /v4l2/camera_info
      param:
      - from: $(find-pkg-share apriltag_ros)/cfg/tags_36h11.yaml
      extra_arg:
      - name: use_intra_process_comms
        value: "True"
```

[... 96 lines cut ...]
[component_container-1] [ERROR] [1685721678.664611317] [v4l2.rectify]: Rectified topic '/v4l2/image_rect' requested but camera publishing '/v4l2/camera_info' is uncalibrated
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/apriltag/apriltag' in container '/apriltag_container'
[component_container-1] [ERROR] [1685721678.665294095] [v4l2.rectify]: Rectified topic '/v4l2/image_rect' requested but camera publishing '/v4l2/camera_info' is uncalibrated
[component_container-1] [ERROR] [1685721678.665885428] [v4l2.rectify]: Rectified topic '/v4l2/image_rect' requested but camera publishing '/v4l2/camera_info' is uncalibrated
```

This error message will then repeat.

Using `ros2 topic list` the `image_rect` can be seen:

```
st@ST:~$ ros2 topic list
/apriltag/detections
/parameter_events
/rosout
/tf
/v4l2/camera_info
/v4l2/image_raw
/v4l2/image_rect
```

Any solution for this issue?
Or did I do something wrong?

Thanks!

christianrauch, 2023-09-01:
The launch file launches three nodes: camera, rectification and the apriltag node. The apriltag node is not involved in the rectification, it just consumes the rectified images. The problem must either be in the package `v4l2_camera` or `image_proc`. If you remove the last section of the launch file with `pkg: apriltag_ros`, you will probably still get the same error.

Having said this, I cannot reproduce your issue. I am using an integrated webcam that I calibrated earlier, and it loads the calibration file without issues.

The message is coming from: https://github.com/ros-perception/image_pipeline/blob/3.0.1/image_proc/src/rectify.cpp#L98-L109 so you might want to check that the intrinsics that are published on `/v4l2/camera_info` actually contain valid values for matrix `k`.

If you still have issues with the calibration, I suggest checking out the repos for the [camera](https://gitlab.com/boldhearts/ros2_v4l2_camera) and [rectification](https://github.com/ros-perception/image_pipeline) node.

closed 2023-09-01 by christianrauch

wangzizhe, 2023-09-11:
I've written a new launch file then this problem disappeared:

```
import launch
from launch_ros.actions import ComposableNodeContainer
from launch_ros.descriptions import ComposableNode

from launch.actions import DeclareLaunchArgument
from launch.actions import IncludeLaunchDescription

from launch.substitutions import LaunchConfiguration
from launch_ros.substitutions import FindPackageShare
from launch.launch_description_sources import PythonLaunchDescriptionSource # to import other launch files

# detect all 36h11 tags
cfg_36h11 = {
    "image_transport": "raw",
    "family": "36h11",
    "size": 0.162,
    "max_hamming": 0,
}

def generate_launch_description():
    camera_resolution_value = LaunchConfiguration('image_size')
    
    camera_resolution_launch_arg = DeclareLaunchArgument(
        'image_size', 
        default_value='[1648,1232]'
    )
    
    image_proc_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([
            FindPackageShare("image_proc"), '/launch/', 'image_proc.launch.py'])
    )
    
    cam_node = ComposableNode(
        package='v4l2_camera',
        # executable='v4l2_camera_node',
        plugin='v4l2_camera::V4L2Camera',
		parameters=[{
            'image_size': LaunchConfiguration('image_size')
        }]
    )
    
    tag_node = ComposableNode(
        name='apriltag_36h11',
        namespace='apriltag',
        package='apriltag_ros', plugin='AprilTagNode',
        remappings=[
                # http://wiki.ros.org/roslaunch/XML/remap
                # This maps the 'raw' images for simplicity of demonstration.
                # ("/apriltag/image_rect", "/camera/image_raw"),
                # In practice, this will have to be the rectified 'rect' images.
                ("/apriltag/image_rect", "/camera/image_rect"),
                ("/apriltag/camera_info", "/camera/camera_info"),
        ],
        parameters=[cfg_36h11],
        extra_arguments=[{'use_intra_process_comms': True}],
    )
        
    container = ComposableNodeContainer(
        name='tag_container',
        namespace='apriltag',
        package='rclcpp_components',
        executable='component_container',
        composable_node_descriptions=[cam_node, tag_node],
        output='screen'
    )

    return launch.LaunchDescription([
[... 87 lines cut ...]
[component_container-1] [WARN] [1685721523.934164798] [v4l2_camera]: Control type not currently supported: 9, for control: Auto Exposure, Bias
[component_container-1] [WARN] [1685721523.934636909] [v4l2_camera]: Control type not currently supported: 9, for control: ISO Sensitivity
[component_container-1] [WARN] [1685721523.935426002] [v4l2_camera]: Control type not currently supported: 6, for control: JPEG Compression Controls
[component_container-1] [INFO] [1685721523.935748409] [v4l2_camera]: Starting camera
[component_container-2] [INFO] [1685721523.963883132] [image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::DebayerNode>
[component_container-2] [INFO] [1685721523.964015261] [image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::DebayerNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/debayer_node' in container '/image_proc_container'
[component_container-2] [INFO] [1685721524.008961613] [image_proc_container]: Load Library: /opt/ros/humble/lib/librectify.so
[component_container-2] [INFO] [1685721524.019409076] [image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[component_container-2] [INFO] [1685721524.019601057] [image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/rectify_mono_node' in container '/image_proc_container'
[component_container-2] [INFO] [1685721524.051258872] [image_proc_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[component_container-2] [INFO] [1685721524.051404446] [image_proc_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_proc::RectifyNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/rectify_color_node' in container '/image_proc_container'
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/v4l2_camera' in container '/apriltag/tag_container'
[component_container-1] [INFO] [1685721524.404695002] [apriltag.tag_container]: Load Library: /home/st/apriltag_ros2_ws/install/apriltag_ros/lib/libAprilTagNode.so
[component_container-1] [INFO] [1685721524.425501131] [apriltag.tag_container]: Found class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[component_container-1] [INFO] [1685721524.425705835] [apriltag.tag_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[component_container-1] [WARN] [1685721524.593323057] [v4l2_camera]: Image encoding not the same as requested output, performing possibly slow conversion: yuv422_yuy2 => rgb8
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/apriltag/apriltag_36h11' in container '/apriltag/tag_container'
[component_container-1] [INFO] [1685721524.639582668] [v4l2_camera]: using default calibration URL
[component_container-1] [INFO] [1685721524.639844946] [v4l2_camera]: camera calibration URL: file:///home/st/.ros/camera_info/mmal_service_16.1.yaml
```

Thanks a lot anyway!