christianrauch/apriltag_ros #20: No detection

Jenanaputra, 2023-10-24:
Hi .. I tried to use the apriltag_ros in ros humble with the input is mono camera. when the package is launched there is no detection. Correct me if iam wrong, therefore,  this package can be used on mono image ?

christianrauch, 2023-10-24:
Do you mean "monocular" (a single camera) or "monochromatic" (grey-scale image) images? Generally, the node subscribes to a single colour topic and does not work with multiple cameras at the same time. I haven't tested grey-scale images, but I don't see why it should not work.

Have you tested the [apriltag library](https://github.com/AprilRobotics/apriltag) on your image data?

jeskesen, 2023-11-18:
I know @Jenanaputra , and asked him if he was still having this problem, and he said it ended up being just user error, so this issue can be closed.   For completeness, he was using **one** of the stereo cameras on an OAK-D, so it was both monochromatic & monocular.

christianrauch, 2023-11-18:
Thanks for getting back on this and clearing this up.

closed 2023-11-18 by christianrauch