christianrauch/apriltag_ros #26: run detector without calibration?

berndpfrommer, 2024-01-16:
I was planning on using the detector for calibration purposes, meaning there is no calibration available yet. I would only be using the corner positions and ID from the detection messages.
Any suggestions how to do this? The only way I see is to add another configuration parameter that would trigger the use of a regular image transport subscription rather than a camera subscription.

christianrauch, 2024-01-23:
You can remap the topic to a topic with uncalibrated images (`image_raw`). You will still need to publish some fake `CameraInfo` messages (e.g. a default constructed message) on `camea_info` to trigger the synchronisation.

berndpfrommer, 2024-01-23:
ok, thanks for clarifying!

closed 2024-01-23 by berndpfrommer