christianrauch/apriltag_ros #28: use realsense d435i, but did not detect anything

yuan-0816, 2024-02-22:
I am using ubuntu 22.04 and ros2 Humble
I want to use infra camera from realsense d435i to detect apriltag (The AprilTag ID in the image is 42)
[image]

It is the grayscale image, I remap the image topic to the apriltag node

but I can not detect anything, did I do something wrong?

here is my launch file

```
import os

from launch import LaunchDescription, Substitution, LaunchContext
from launch.actions import DeclareLaunchArgument, SetEnvironmentVariable, LogInfo, OpaqueFunction
from launch.substitutions import LaunchConfiguration, ThisLaunchFileDir, PythonExpression
from launch.conditions import IfCondition, UnlessCondition
from launch_ros.actions import Node
from launch_ros.actions import SetParameter
from typing import Text
from ament_index_python.packages import get_package_share_directory

def launch_setup(context, *args, **kwargs):     
    
    apriltag_yaml = os.path.join(
        get_package_share_directory('apriltag_ros'), 'apriltag_ros', 'cfg', 'tags_36h11.yaml')   

    return [

        
        Node(
            package='apriltag_ros', executable='apriltag_node', output='screen',
            remappings=[
                ('/apriltag/image_rect', '/camera/infra1/image_rect_raw'),
                ('/apriltag/camera_info', '/camera/infra1/camera_info')
            ],
            parameters=[{'from': apriltag_yaml}],
            # extra_arguments=['use_intra_process_comms:=True']
        )

          
    ]

def generate_launch_description():
    

    return LaunchDescription([
        OpaqueFunction(function=launch_setup)

    ])
```

and my yaml file 
```
apriltag:
    ros__parameters:
        image_transport: raw    # image format
        family: 36h11           # tag family name
        size: 0.160             # tag edge size in meter
        max_hamming: 0          # maximum allowed hamming distance (corrected bits)

        # see "apriltag.h" 'struct apriltag_detector' for more documentation on these optional parameters
        detector:
            threads: 1          # number of threads
            decimate: 2.0       # decimate resolution for quad detection
            blur: 0.0           # sigma of Gaussian blur for quad detection
            refine: True        # snap to strong gradients
            sharpening: 0.25    # sharpening of decoded images
            debug: True        # write additional debugging images to current working directory

        # optional list of tags
        tag:
            ids: [42]               # tag ID
            frames: [base, object]  # optional frame name
            sizes: [0.160, 0.160]   # optional tag-specific edge size
```

christianrauch, 2024-02-22:
Please check with the apriltag_demo from the apriltag library first before reporting such issues to the ROS node.

Looking at the image, my best guess is that your fingers are interfering with the edge detection.

yuan-0816, 2024-02-28:
I can detect using your launch file, but I want to use Python to write the launch file. However, I'm not sure how to implement the following parameters in Python

```
    - pkg: apriltag_ros
      plugin: AprilTagNode
      name: apriltag
      namespace: apriltag
      remap:
      - from: /apriltag/image_rect
        to: /v4l2/image_rect
      - from: /apriltag/camera_info
        to: /v4l2/camera_info
      param:
      - from: $(find-pkg-share apriltag_ros)/cfg/tags_36h11.yaml
      extra_arg:
      - name: use_intra_process_comms
        value: "True"
```

christianrauch, 2024-02-28:
> I can detect using your launch file, but I want to use Python to write the launch file.

That's a bit out of scope for this issue and this repo is not the best place to ask common ROS questions.

Just googling for "ros2 launch files parameters" gives me:
- https://roboticscasual.com/tutorial-ros2-launch-files-all-you-need-to-know/
- https://roboticsbackend.com/ros2-launch-file-example/
- https://docs.ros.org/en/foxy/How-To-Guides/Launch-file-different-formats.html

If this is still unclear or you have a very specific problem with launch files, then I suggest asking your question at https://robotics.stackexchange.com.

You mentioned that your original issue with the tag detection has been solved. I am going to close this issue.

closed 2024-02-28 by christianrauch

yuan-0816, 2024-02-29:
ok, I will try another method. 
thanks for your reply><