christianrauch/apriltag_ros #37: Orientation difference between pnp and homography

eliasdc, 2024-06-27:
The pose estimations have a different resulting frame.

- homography: z points inwards into the tag
- pnp: z points out of the tag

christianrauch, 2024-06-30:
I can reproduce this and am surprised that this wasn't caught in the discussions in https://github.com/christianrauch/apriltag_ros/issues/14. The z-axis of the tag frame is supposed to point away from the tag towards the camera (a.k.a. "outwards") as it is with the `pnp` option, which has been adapted from the ROS 1 node. See the second half of https://github.com/christianrauch/apriltag_ros/issues/14#issuecomment-1883943870 for an explanation of the tag frame orientation.

The `homography` option should produce the same tag frame orientation. IMHO, adding another option for the preferred tag orientation is not so useful as users can always apply additional transformations on top to achieve their desired tag orientation.

christianrauch, 2024-07-02:
Does https://github.com/christianrauch/apriltag_ros/pull/38 fix your issue?

PR #38 (2024-07-07): fix orientation of tag frame estimated by homography method
Fixes #37 .

Frame comparison for the same input sequence:

| pnp | homography |
| - | - |
| <video src="https://github.com/christianrauch/apriltag_ros/assets/8573621/02018c73-4a02-48cd-aab3-22881c3d9338"> | <video src="https://github.com/christianrauch/apriltag_ros/assets/8573621/c12eb212-374b-4252-87d4-5635ea14267a"> |
closed 2024-07-07 by christianrauch