christianrauch/apriltag_ros #39: No apriltag detections

izeemeen, 2024-07-30:
Hi there!
I launch the camera from the following git:
**https://github.com/christianrauch/camera_ros**

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-30-10-36-58-832345-ubuntu-2701
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [2713]
[component_container-1] [INFO] [1722335819.486322921] [camera_container]: Load Library: /home/ubuntu/camera_ws/install/camera_ros/lib/libcamera_component.so
[component_container-1] [INFO] [1722335819.712739952] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container-1] [INFO] [1722335819.712916228] [camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<camera::CameraNode>
[component_container-1] [1:40:12.003768545] [2713]  INFO Camera camera_manager.cpp:313 libcamera v0.3.0+158-929978ee
[component_container-1] [1:40:12.050168992] [2724]  WARN RPiSdn sdn.cpp:40 Using legacy SDN tuning - please consider moving SDN inside rpi.denoise
[component_container-1] [1:40:12.054007213] [2724]  INFO RPI vc4.cpp:400 Registered camera /base/soc/i2c0mux/i2c@1/ov5647@36 to Unicam device /dev/media0 and ISP device /dev/media1
[component_container-1] [1:40:12.055353977] [2713]  INFO Camera camera.cpp:1183 configuring streams: (0) 1280x720-YUYV
[component_container-1] [1:40:12.056114914] [2724]  INFO RPI vc4.cpp:569 Sensor: /base/soc/i2c0mux/i2c@1/ov5647@36 - Selected sensor format: 1920x1080-SGBRG10_1X10 - Selected unicam format: 1920x1080-pGAA
[component_container-1] [INFO] [1722335819.797299206] [camera]: camera "/base/soc/i2c0mux/i2c@1/ov5647@36" configured with 1280x720-YUYV stream
[component_container-1] [WARN] [1722335819.800135159] [camera]: control NoiseReductionMode (10002) not handled
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera' in container '/camera_container'
[component_container-1] [INFO] [1722335819.925118993] [camera_container]: Load Library: /opt/ros/humble/lib/libimage_view_nodes.so
[component_container-1] [INFO] [1722335820.089359845] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_view::DisparityViewNode>
[component_container-1] [INFO] [1722335820.089540732] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_view::ExtractImagesNode>
[component_container-1] [INFO] [1722335820.089577843] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_view::ImageSaverNode>
[component_container-1] [INFO] [1722335820.089604657] [camera_container]: Found class: rclcpp_components::NodeFactoryTemplate<image_view::ImageViewNode>
[component_container-1] [INFO] [1722335820.089633027] [camera_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<image_view::ImageViewNode>
[component_container-1] [INFO] [1722335820.110605930] [image_view_node]: Using transport "raw"
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/image_view_node' in container '/camera_container'
[component_container-1] [INFO] [1722335820.241191709] [camera]: using default calibration URL
[component_container-1] [INFO] [1722335820.241504743] [camera]: camera calibration URL: file:///home/ubuntu/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1280x720.yaml
[component_container-1] [ERROR] [1722335820.241811055] [camera_calibration_parsers]: Unable to open camera calibration file [/home/ubuntu/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1280x720.yaml]
[component_container-1] [WARN] [1722335820.241911776] [camera]: Camera calibration file /home/ubuntu/.ros/camera_info/ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_1280x720.yaml not found

After that I launch apriltag node:

[INFO] [launch]: All log files can be found below /home/ubuntu/.ros/log/2024-07-30-10-37-11-629815-ubuntu-2740
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [component_container-1]: process started with pid [2752]
[component_container-1] [INFO] [1722335832.753609931] [apriltag_container]: Load Library: /opt/ros/humble/lib/libv4l2_camera.so
[component_container-1] [INFO] [1722335833.048811615] [apriltag_container]: Found class: rclcpp_components::NodeFactoryTemplate<v4l2_camera::V4L2Camera>
[component_container-1] [INFO] [1722335833.049018854] [apriltag_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<v4l2_camera::V4L2Camera>
[component_container-1] [INFO] [1722335833.152663233] [v4l2_camera]: Driver: unicam
[component_container-1] [INFO] [1722335833.154444679] [v4l2_camera]: Version: 331678
[component_container-1] [INFO] [1722335833.154486623] [v4l2_camera]: Device: unicam
[component_container-1] [INFO] [1722335833.154505641] [v4l2_camera]: Location: platform:fe801000.csi
[component_container-1] [INFO] [1722335833.154524863] [v4l2_camera]: Capabilities:
[component_container-1] [INFO] [1722335833.154543770] [v4l2_camera]:   Read/write: YES
[component_container-1] [INFO] [1722335833.154561085] [v4l2_camera]:   Streaming: YES
[component_container-1] [INFO] [1722335833.154602195] [v4l2_camera]: Current pixel format: pGAA @ 1920x1080
[component_container-1] [INFO] [1722335833.154675658] [v4l2_camera]: Available pixel formats: 
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/camera/camera' in container '/apriltag_container'
[component_container-1] [INFO] [1722335833.154698657] [v4l2_camera]: Available controls: 
[component_container-1] [INFO] [1722335833.159269575] [v4l2_camera]: Requesting format: 1920x1080 YUYV
[ERROR] [launch_ros.actions.load_composable_nodes]: Failed to load node 'rectify' of type 'image_proc::RectifyNode' in container '/apriltag_container': Could not find requested resource in ament index
[component_container-1] [ERROR] [1722335833.159446610] [v4l2_camera]: Failed requesting pixel format: Device or resource busy (16)
[component_container-1] [INFO] [1722335833.159488258] [v4l2_camera]: Requesting format: 640x480 pGAA
[component_container-1] [ERROR] [1722335833.159518887] [v4l2_camera]: Failed requesting pixel format: Device or resource busy (16)
[component_container-1] [INFO] [1722335833.163751605] [v4l2_camera]: Starting camera
[component_container-1] [ERROR] [1722335833.163910844] [v4l2_camera]: Failed mapping device memory
[component_container-1] [ERROR] [1722335833.173678210] [apriltag_container]: Could not find requested resource in ament index
[component_container-1] [INFO] [1722335833.180518068] [apriltag_container]: Load Library: /home/ubuntu/colcon_ws/install/apriltag_ros/lib/libAprilTagNode.so
[component_container-1] [INFO] [1722335833.212019200] [apriltag_container]: Found class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[component_container-1] [INFO] [1722335833.212182699] [apriltag_container]: Instantiate class: rclcpp_components::NodeFactoryTemplate<AprilTagNode>
[INFO] [launch_ros.actions.load_composable_nodes]: Loaded node '/apriltag/apriltag' in container '/apriltag_container'
[component_container-1] [WARN] [1722335834.253500095] [apriltag.apriltag]: [image_transport] Topics '/camera/image_rect' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
[component_container-1] 	Image messages received:      0
[component_container-1] 	CameraInfo messages received: 30
[component_container-1] 	Synchronized pairs:           0
[component_container-1] [WARN] [1722335835.253314180] [apriltag.apriltag]: [image_transport] Topics '/camera/image_rect' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
[component_container-1] 	Image messages received:      0
[component_container-1] 	CameraInfo messages received: 30
[component_container-1] 	Synchronized pairs:           0
[... 7 lines cut ...]
[component_container-1] 	Synchronized pairs:           0
[component_container-1] [WARN] [1722335838.253192935] [apriltag.apriltag]: [image_transport] Topics '/camera/image_rect' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
[component_container-1] 	Image messages received:      0
[component_container-1] 	CameraInfo messages received: 31
[component_container-1] 	Synchronized pairs:           0
[component_container-1] [WARN] [1722335839.253378611] [apriltag.apriltag]: [image_transport] Topics '/camera/image_rect' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:
[component_container-1] 	Image messages received:      0
[component_container-1] 	CameraInfo messages received: 30
[component_container-1] 	Synchronized pairs:           0

My **ros2 topic list**:

/apriltag/detections
/camera/camera_info
/camera/image_raw
/camera/image_raw/compressed
/camera/image_rect
/output
/parameter_events
/rosout
/tf

**ros2 topic echo /apriltag/detections** command doesn't give me anything

Where can I make a mistake?

christianrauch, 2024-07-30:
The issue is in `Topics '/camera/image_rect' and '/camera/camera_info' do not appear to be synchronized. In the last 10s:`. Did you check that the topics are correctly set and that they publish messages?

izeemeen, 2024-07-30:
Thanks for answer @christianrauch!
Can you help me where can I find `RectifyNode`?

christianrauch, 2024-07-30:
> Can you help me where can I find `RectifyNode`?

Please use the public ROS documentation and search engines to look for this kind of information. Simply searching for "ros2 RectifyNode" in a very popular search engine will give you already a hint of where to find it and which Debian package you have to install. The same goes for introspecting the ROS network and the calibration.

closed 2024-08-06 by izeemeen