christianrauch/apriltag_ros #40: Is Synchronization  between camera_info and camera topic necessary

SamuelHafner, 2024-08-14:
Hi @christianrauch,

thanks very much for your amazing work in different ros2 packages. 

I wanted to ask, why is the synchronization necessary? Because the camera info information will not change during runtime and so we only need the camera info information (camera matrix etc.) once. 

Thank you

christianrauch, 2024-08-16:
> I wanted to ask, why is the synchronization necessary? Because the camera info information will not change during runtime and so we only need the camera info information (camera matrix etc.) once.

That depends on your camera. E.g. if your camera has an optical zoom that changes at runtime, your intrinsics have to adapt.  Cameras also have the option to be recalibrated at runtime after which they will/should publish the new calibrated intrinsics. I haven't used such cameras myself yet and I also think that changing intrinsics at runtime is quite uncommon. But nevertheless, this is how the interface is rightfully defined and the `image_transport::CameraPublisher` for example provides functions to publish the camera image and the camera info at the same time.

thuhangkhuat, 2024-10-09:
Hi @christianrauch. 
Thanks for your apriltag_ros package in ROS2. I have a question. Could the data type float64[9] homography (3x3 row-major homography matrix) in apriltag_msgs replace the quaternion data in the message of ROS 1? 
Thank you.

SamuelHafner, 2024-10-17:
Okay perfect. That makes sense.

Thank you very much.

closed 2024-10-17 by SamuelHafner

NikolasE, 2024-12-19:
I'm running into the same problem while streaming images and camera_info from IsaacSim, so I can't change the publisher. To still be able to use the data, I added a flag into this package to use normal subscribers and read the camera_info only once. 
Would you merge such a PR? 

In this case, the user would actively announce that the intrinsics won't change (and we could even verify this for all camera_info message that the node receives).

josephduchesne, 2025-05-24:
I'm also running into this issue with depthai cameras. I'm not sure if it's a CPU load issue, or a bug in depthai, image transport or apil tag node, but it falls slowly out of sync in a weird way (after a while I get 1Hz rather than 10Hz updates from april tag node, but am still seeing 10Hz from the camera driver. In any case, it's a fixed focus camera (like the vast majority of cameras used to detect april tags in ROS2 I'm betting), so synchronizing isn't needed at all.

christianrauch, 2025-05-25:
Like many other nodes, the apriltag node is just using [`image_transport::CameraSubscriber`](https://github.com/ros-perception/image_common/blob/913f5acae4ebd45e904dbb03d1f9086472b580f6/image_transport/include/image_transport/camera_subscriber.hpp#L63), and this in turn is using a [`message_filters::TimeSynchronizer<Image, CameraInfo>`](https://github.com/ros-perception/image_common/blob/913f5acae4ebd45e904dbb03d1f9086472b580f6/image_transport/src/camera_subscriber.cpp#L52) to synchronise images and intrinsics. The `TimeSynchronizer` is essentially expecting that all messages are stamped and have the exact same time stamp. If the camera node for your camera cannot provide this, you should open an issue there to investigate why the nodes does not publish the image and intrinsics with matching timestamps. In principle, replacing the `CameraSubscriber` with something else in this node just shifts the issue. There are many more users of `CameraSubscriber`, such as the standard rectification node, and you will have a hard time convincing everybody else to change this, just because the ROS node provided to you does not behave as expected.

In principle, I am not willing to replace `CameraSubscriber` just to work around an issue in someone else's camera node. But what you can try, and what I would consider as a PR, is to use a different QoS profile for the `CameraSubscriber`. I am using `rmw_qos_profile_sensor_data`:
```C++
static const rmw_qos_profile_t rmw_qos_profile_sensor_data =
{
  RMW_QOS_POLICY_HISTORY_KEEP_LAST,
  5,
  RMW_QOS_POLICY_RELIABILITY_BEST_EFFORT,
  RMW_QOS_POLICY_DURABILITY_VOLATILE,
  RMW_QOS_DEADLINE_DEFAULT,
  RMW_QOS_LIFESPAN_DEFAULT,
  RMW_QOS_POLICY_LIVELINESS_SYSTEM_DEFAULT,
  RMW_QOS_LIVELINESS_LEASE_DURATION_DEFAULT,
  false
};
```
as this is the most appropriate for a live camera feed IMHO, whereas the default for `image_transport::create_camera_subscription` would be `rmw_qos_profile_default`:
```C++
static const rmw_qos_profile_t rmw_qos_profile_default =
{
  RMW_QOS_POLICY_HISTORY_KEEP_LAST,
  10,
  RMW_QOS_POLICY_RELIABILITY_RELIABLE,
  RMW_QOS_POLICY_DURABILITY_VOLATILE,
  RMW_QOS_DEADLINE_DEFAULT,
  RMW_QOS_LIFESPAN_DEFAULT,
  RMW_QOS_POLICY_LIVELINESS_SYSTEM_DEFAULT,
  RMW_QOS_LIVELINESS_LEASE_DURATION_DEFAULT,
  false
};
```

This mainly changes the history depth from 5 to 10 and changes the reliability from "BEST_EFFORT" to "RELIABLE". You can play around with either settings or replace the QoS profile entirely to see if this helps in your case. If you find this useful, I would accept a PR to have the QoS profile configurable via a parameter.

cheriehu, 2025-11-10:
I was also facing issues with synchronization and got warnings saying that the camera_info and image_raw topics were not synced. Changing the QOS profile to `rmw_qos_profile_default` does seem to help for me, the warning doesn't appear. Would it be possible to change the QOS profile for the camera subscriber to `rmw_qos_profile_default` instead of `rmw_qos_profile_sensor_data` for this package?

christianrauch, 2025-11-10:
> Would it be possible to change the QOS profile for the camera subscriber to `rmw_qos_profile_default` instead of `rmw_qos_profile_sensor_data` for this package?

This could be a `qos_profile` parameter mapping from strings to profile enums, e.g. `"default"` -> `rmw_qos_profile_default`, `"sensor_data"` -> `rmw_qos_profile_sensor_data`, ..., with `rmw_qos_profile_default` as the default setting. I would review a PR configuring the `CameraSubscriber` this way.