christianrauch/apriltag_ros #41: GoPiGo3 Robot Dave "Sees" Using Your Wrapper

slowrunner, 2024-12-19:
Not an issue - but you don't have a Discussions section so I'm abusing your site - sorry.

Christian, long ago you were the first to put ROS on the GoPiGo3 platform.  My ROS 2 Humble GoPiGo3 robot "GoPi5Go-Dave" is gearing up to use Steve Macenski's new Nav2 docking feature, so I'm trying out your apriltag_ros wrapper.  

Dave has the wide FOV Luxonis Oak-D-W97 camera so I was wondering just how far away he would be able to detect a dock tag.  Tonight I managed to get it installed and listening to the camera.  (I've been learning ROS 2 for the last couple years, and Dave is a 24/7 ROSbot getting on and off his dock, but cannot "find his way home" yet so he only gets off 17cm and then back on 17cm.)

This is GoPi5Go-Dave:
[image]

And tonight he detected a Standard 36h11 tag (150mm on a side) from about 6 meters.  I ran the apriltag_node and see /detections topic - have to figure out how to get the tag pose if possible.

[image]

christianrauch, 2024-12-19:
Hm, do you have a particular issue with ROS node or do you just wanted to share this here? In your second image, the tag is quite far away. I would be surprised if it gets detected, even more if the pose would be correct. You could simplify this, by just taking images and testing out different parameters with the standalone apriltag examples.

slowrunner, 2024-12-19:
Just sharing - thank you.

closed 2024-12-19 by slowrunner