christianrauch/apriltag_ros #42: ROS2 and AprilTag

mauricefallon, 2025-02-05:
Hi @christianrauch,

It's been a while - I hope you are doing well!

I picked up the ROS2 April Tag thread here:
https://github.com/AprilRobotics/apriltag_ros/issues/76
I know you and Wolf are 'core developers' of AprilTags.

So in brief there is no 'official ROS2 port of the application. Is that right?
In which case, would you recommend your branch - (obviously biased!)

(I know the professor of that lab runs a company now - Ed Olson)

christianrauch, 2025-02-05:
Hi @mauricefallon, nice to meet you here.

I created that ROS 2 node many years ago with the intention to move this to https://github.com/AprilRobotics eventually. If someone adds me to @AprilRobotics, I can move the repo there. My ROS 2 node only supports the tag detection and pose estimation for individual tags for a stream of images on a topic. I did not port all the feature of the ROS 1 node, such as tag bundles or the detection service for individual images. Also, I am not using the node myself at the moment, so all maintenance is "best effort".

As an alternative and still maintained repo, I am only aware of https://github.com/ros-misc-utilities/apriltag_detector. But this node does not seem to compute and publish the transformations.

In summary: If you need the individual transforms of tags, I would recommend my node. But if you need more features, such as a detection service or tag bundles, someone needs to implement it for this `apriltag_ros` repo or work on an independent 1:1 port of the original ROS 1 node (https://github.com/AprilRobotics/apriltag_ros).

closed 2025-02-05 by mauricefallon

mauricefallon, 2025-02-11:
Hi @christianrauch ,
Can you explain why you created an entirely different msg type for AprilTagDetection.msg ?

The ROS1 version contains just the detected pose. While the ROS2 contains 'raw' detection data:
[image]

christianrauch, 2025-02-11:
When I started the node from scratch I did not port the ROS 1 node and messages directly to ROS 2. For visualisation and debugging, I created a message type that contains the raw 2D tag detection only. The 3D pose of the tag is already published as a TF.

Do you see a need for a dedicated pose message? If this is useful for others, I would suggest adding a new message definition `AprilTagPose.msg` to distinguish the 2D detection and 3D pose.

mauricefallon, 2025-02-11:
I see - its just unfortunate that you replicated exactly the same message file names.

Its useful to have the detections as an organised set (e.g. for doing tag-based navigation), rather than in the TF message stream.

But I can see that the node can be easily modified to publish `AprilTagPoseArray.msg` right here:
https://github.com/christianrauch/apriltag_ros/blob/master/src/AprilTagNode.cpp#L229

I'll make a PR if/when I make those changes. I'll adapt the older ROS! AprilTagDetection.msg

thanks!