christianrauch/apriltag_ros #43: Do /tf TransformStamped messages really contain a tag pose

billmania, 2025-03-03:
I found the line "The tag poses are published on the standard TF topic /tf with the header set to the image header and child_frame_id set to either tag<family>:<id> (e.g. "tag36h11:0") or the frame name selected via configuration file." in this repo's README. However, in some initial experimentation, I'm not getting a reasonable-looking position vector in the transforms translation vector.

Is it really supposed to be a Pose, shoved into a Transform? Why not just publish a Pose instead?

I'm running ROS2 Humble with version 3.2.2 of this repo. I have 36h11 Tag ID 586, printed as a 15.2 cm square. The tags are detected and the centre.x and centre.y change reasonably, as I move the tag parallel to the plane of the camera sensor. At the same time, I notice the transform.translation.x and .y change very little, even though I move the tag about a meter along both the x and y axes.

I've set pose_estimation_method to both "pnp" and "homography", but didn't notice any difference in the translation vector.

As examples, detections contains:
```
detections:
- family: tag36h11
  id: 586
  hamming: 0
  goodness: 0.0
  decision_margin: 94.36902618408203
  centre:
    x: 325.77281293622326
    y: 303.95897009607216
  corners:
  - x: 305.09518432617193
    y: 324.10260009765625
  - x: 345.72875976562506
```
while tf contains:
```
- header:
    stamp:
      sec: 1741017247
      nanosec: 620389966
    frame_id: camera
  child_frame_id: six
  transform:
    translation:
      x: -0.09888039876294519
      y: 0.2474343915432532
      z: 4.783722634230373
    rotation:
      x: -0.9894874576775022
      y: -0.017557024763264786
      z: -0.0393001362593627
      w: 0.13806455472122747
```

christianrauch, 2025-03-03:
> However, in some initial experimentation, I'm not getting a reasonable-looking position vector in the transforms translation vector.

1. Can you test this with the apriltag library too and show the results here? The library has similar functions for the pose estimation.
2. Can you provide a log file etc. to reproduce this?

> Is it really supposed to be a Pose, shoved into a Transform? Why not just publish a Pose instead?

The pose of the tag in the camera frame is given by the transformation from the camera frame to the tag frame. A `TransformStamped` is more specific than a `PoseStamped` as it defines the parent and the child frame and, hence, allows you to use the node directly to connect different TF trees. The content of the `transform.translation` and `pose.position` respectively, and `transform.rotation` and `pose.orientation` respectively, would be the same. So I don't see how this would be related to the effect you are describing.

Do you specifically need the `PoseStamped` message type in addition to the `TransformStamped` anyway? This was requested before (https://github.com/christianrauch/apriltag_ros/issues/42#issuecomment-2650248963) and I suggested adding a new message definition `AprilTagPose.msg` / `AprilTagPoseArray.msg` for those that want to subscribe to the poses instead of using the TFs.

billmania, 2025-03-03:
For my application, I can use either the TransformStamped or a PoseStamped (or even just a Pose). For my Use Case, I'm determining only whether the APRIL tag is centered horizontally in the camera frame, ie. close to width/2, and the distance from the camera to the tag.

By "log file", do you mean a rosbag containing both /detections and /tf? I see only two DEBUG log statements in AprilTagNode.cpp, so I'm assuming that wouldn't give much useful information.

By "test this with the apriltag library" do you mean ignore the content of the TransformStamped message and instead pass the data from the /detections topic to the appropriate apriltag library function in my own code?

There are (at least) two other areas where I'm uncertain. The only two coordinate frames in my application are "camera" and "six". I'm not publishing any static transforms from world or base_link to camera. Is that an issue? And, if transform.translation is equivalent to pose.position, are meters the units of transform.translation?

christianrauch, 2025-03-04:
> By "log file", do you mean a rosbag containing both /detections and /tf? I see only two DEBUG log statements in AprilTagNode.cpp, so I'm assuming that wouldn't give much useful information.

Of course, I need the original data to reproduce this. So either the original bag file or a simulation would do it.

> By "test this with the apriltag library" do you mean ignore the content of the TransformStamped message and instead pass the data from the /detections topic to the appropriate apriltag library function in my own code?

Yes. I mean testing your input data with https://github.com/AprilRobotics/apriltag. You probably need to export one of the images from the topic as `jpeg` or `pnm` file, run the detection and then use the rest of the apriltag API to do the pose estimation with your intrinsics and tag size. This is always to see if there is an issue cause by the ROS node or if this is already in the library.

> There are (at least) two other areas where I'm uncertain. The only two coordinate frames in my application are "camera" and "six". I'm not publishing any static transforms from world or base_link to camera. Is that an issue?

I am not sure. Depends on what you want to do. But the TF is what it is. It is published on the "global" `/tf` topic, so every node can use the TF API. You can just query the transform from the camera frame to the tag frame, if you do not have any other frames.

> And, if transform.translation is equivalent to pose.position, are meters the units of transform.translation?

Yes. Both are given in meters. But these values only make sense if your camera node is publishing the correct intrinsics. Is your camera calibrated?

billmania, 2025-03-05:
I've created a 614 MB compressed tar file which contains:

- a ROS bag with the /tf, /detections, camera_info, and image_rect topics
- the camera calibration yaml
- the apriltag_ros configuration yaml

Before I try to push a file of that size to github, where would you like me to place this large tar file?

christianrauch, 2025-03-05:
>     * a ROS bag with the /tf, /detections, camera_info, and image_rect topics

Which camera driver are you using? Could you log the "compressed" images instead of the raw (uncompressed) images? I think it would also be sufficient to have a short 5 seconds snipped, or so, of the time when you get the estimated pose in question.

>     * the camera calibration yaml

I think I do not need this. The camera driver should already publish the correct intrinsics and this is anyway only useful if I run the same camera.

>     * the apriltag_ros configuration yaml

> Before I try to push a file of that size to github, where would you like me to place this large tar file?

I would first try to reduce the size further by compressing the image messages and shorten the bag duration.

Can you further just share an image file here with the intrinsics and the tag settings, so this can be tested with the apriltag library without ROS?

billmania, 2025-03-10:
Sorry, got distracted for a few days.

There's a 3.2 MB compressed tar archive at http://www.manialabs.us/apriltag_ros_issue43.tgz  It contains a db3 data file and the corresponding metadata.yaml. There are a total of 74 messages, from /rq_camera_node0/image_raw/compressed and /rq_camera_node0/camera_info.

The tag 36h11 ID 586 is about 120 cm from the camera.

christianrauch, 2025-03-10:
Thanks for the log. With this small size, you can actually just upload it directly here in the GitHub issue, but you may have to compress this as zip instead.

I am playing back your log file with remapping:
```sh
ros2 bag play apriltag_ros_issue43 --remap /rq_camera_node0/camera_info:=/camera/camera/camera_info /rq_camera_node0/image_raw/compressed:=/camera/camera/image_raw/compressed -l
```
and use the following launch file with `image_proc::RectifyNode` and `AprilTagNode`:
```yaml
launch:
- node_container:
    pkg: rclcpp_components
    exec: component_container
    name: apriltag_container
    namespace: ""
    composable_node:
    - pkg: image_proc
      plugin: image_proc::RectifyNode
      name: rectify
      namespace: camera
      remap:
      - from: image
        to: /camera/camera/image_raw
      - from: camera_info
        to: /camera/camera/camera_info
      param:
      - name: image_transport
        value: compressed
      extra_arg:
      - name: use_intra_process_comms
        value: "True"

    - pkg: apriltag_ros
      plugin: AprilTagNode
      name: apriltag
      namespace: apriltag
      remap:
      - from: /apriltag/image_rect
        to: /camera/image_rect
      - from: /camera/camera_info
        to: /camera/camera/camera_info
      param:
      - from: $(find-pkg-share apriltag_ros)/cfg/tags_36h11.yaml
      extra_arg:
      - name: use_intra_process_comms
        value: "True"
```
with this configuration:
```yaml
/**:
    ros__parameters:
        image_transport: raw    # image format
        family: 36h11           # tag family name
        size: 0.152             # tag edge size in meter
        max_hamming: 0          # maximum allowed hamming distance (corrected bits)

        # see "apriltag.h" 'struct apriltag_detector' for more documentation on these optional parameters
        detector:
            threads: 1          # number of threads
            decimate: 2.0       # decimate resolution for quad detection
            blur: 0.0           # sigma of Gaussian blur for quad detection
            refine: True        # snap to strong gradients
            sharpening: 0.25    # sharpening of decoded images
            debug: False        # write additional debugging images to current working directory

        pose_estimation_method: "pnp"   # method for estimating the tag pose
```

On topic `/apriltag/detections` I am getting:
```yaml
header:
  stamp:
    sec: 1741639575
    nanosec: 220003516
  frame_id: camera
detections:
- family: tag36h11
  id: 586
  hamming: 0
  goodness: 0.0
  decision_margin: 118.55689239501953
  centre:
    x: 399.12599877239654
    y: 275.83814979449323
  corners:
  - x: 379.5143737792969
    y: 295.0056152343749
  - x: 418.9411926269531
    y: 296.8807067871093
  - x: 419.9059753417969
    y: 255.52879333496102
  - x: 379.3585815429688
    y: 254.84632873535162
  homography:
  - 13.971734490149911
  - 5.3377929679001594
  - 399.12599877239654
  - -3.5082314649740782
  - 24.192450342753546
  - 275.83814979449323
  - -0.015066308518442194
  - 0.013859298070661283
  - 1.0
```
and on `/tf`:
```
transforms:
- header:
    stamp:
      sec: 1741639575
      nanosec: 187978516
    frame_id: camera
  child_frame_id: tag36h11:586
  transform:
    translation:
      x: 0.2111365874045122
      y: 0.16561172602314972
      z: 4.752241846374453
    rotation:
      x: -0.9873866330155907
      y: -0.007249001164251372
      z: 0.15648539285338764
      w: 0.022965425051082074
```

According to this, the tag should indeed be `4.75` m away from the camera.

The orientation of the tag
```
[  0.9525029, -0.0060894, -0.3044687;
   0.0060894, -0.9992193,  0.0390344;
  -0.3044687, -0.0390344, -0.9517222 ]
```
seems to match.

christianrauch, 2025-03-10:
Which camera are you using and how did you calibrate it? The rectified images still look quite distorted on the edges.

According to the detection, the tag is roughly 39 pixel wide, corresponding to the 15.2 cm. That means, at a focal length (fx) of 1296 pixels, the tag should be 5.05 meters away (0.152 m / 39 pxl *1296 pxl). So, if this is really only 1.2 meters away from the camera, then the focal length should be around 307 pixel (39 pxl / 0.152 m * 1.2 m). This is just roughly estimated, assuming the tag edges are parallel to the image edges. But your focal length is much higher than expected.

billmania, 2025-03-11:
@christianrauch I'm using an "Arducam USB Camera: Arducam USB", with lsusb details "0c45:6366 Microdia Webcam Vitade AF". I created the calibration pattern according to https://calib.io/pages/camera-calibration-pattern-generator and then followed the calibration process as described in https://docs.ros.org/en/ros2_packages/jazzy/api/camera_calibration/doc/tutorial_mono.html.
Should I find a slightly higher-quality camera, calibrate it, and then run the same test again?

christianrauch, 2025-03-11:
> I'm using an "Arducam USB Camera: Arducam USB", with lsusb details "0c45:6366 Microdia Webcam Vitade AF".

There are different variants of the Arducam and I cannot find anything related to "Arducam" and "Microdia Webcam Vitade AF". Which variant do you have? I would roughtly compare the FoV of the camera with you calibration values to see if they are in the same ballpark. From the focal lengths in the bag file, the FoV should be around 28° x 21°. That appears very small to me compared to other cameras. Also, some of the modules support auto-focus and lense mounts for lenses with variable focus. Does the "AF" stand for auto-focus? I would just make sure to keep the focus fixed during and after calibration.

> Should I find a slightly higher-quality camera, calibrate it, and then run the same test again? If yes, give me some guidance on which characteristics make an optimal camera for APRIL tag detection, when the tags are between 1 meter and 10 meters from the camera.

Any camera should do. But you need rectified images for the detection and the camera intrinsics for the 3D pose estimation. On top of that, the higher the resolution the higher the accuracy becomes. You can start with a simple HD or 4k USB webcam.

christianrauch, 2025-03-11:
For reference, this is the `CameraInfo` message that I read from your bag file:
```yaml
header:
  stamp:
    sec: 1741639574
    nanosec: 631898516
  frame_id: camera
height: 480
width: 640
distortion_model: plumb_bob
d:
- -1.831894
- 8.113775
- -0.061439
- 0.078607
- 0.0
k:
- 1415.46001
- 0.0
- 318.49754
- 0.0
- 1349.18889
- 241.53498
- 0.0
- 0.0
- 1.0
r:
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
- 0.0
- 0.0
- 1.0
p:
- 1296.02197
- 0.0
- 341.73914
- 0.0
- 0.0
- 1277.72705
- 231.04377
- 0.0
- 0.0
- 0.0
- 1.0
- 0.0
binning_x: 0
binning_y: 0
roi:
  x_offset: 0
  y_offset: 0
  height: 0
  width: 0
  do_rectify: false
```

billmania, 2025-03-11:
That's correct. The values came from this yaml file:
```
image_width: 640
image_height: 480
camera_name: ArducamUSBCamera_ArducamUSB__base_scb_pcie_7d500000_pci_0_0_usb_0_0_1_1_1_0_0c45_6366_640x480
camera_matrix:
  rows: 3
  cols: 3
  data: [1415.46001,    0.     ,  318.49754,
            0.     , 1349.18889,  241.53498,
            0.     ,    0.     ,    1.     ]
distortion_model: plumb_bob
distortion_coefficients:
  rows: 1
  cols: 5
  data: [-1.831894, 8.113775, -0.061439, 0.078607, 0.000000]
rectification_matrix:
  rows: 3
  cols: 3
  data: [1., 0., 0.,
         0., 1., 0.,
         0., 0., 1.]
projection_matrix:
  rows: 3
  cols: 4
  data: [1296.02197,    0.     ,  341.73914,    0.     ,
            0.     , 1277.72705,  231.04377,    0.     ,
            0.     ,    0.     ,    1.     ,    0.     ]
```

billmania, 2025-03-11:
I found another camera which identifies as "ov5647__base_soc_i2c0mux_i2c_1_ov5647_36_640x480.yaml" when connected to the camera port on the Raspberry Pi 4B. Yes, I have a lot of cheap cameras lying around. This one doesn't have any markings on it, but it does have a much larger lens, and looks to be higher quality, than the original ArduCam. I'm in the process of creating a calibration file for it and will then detect some APRIL tags.

billmania, 2025-03-13:
I'm not certain the ov5647 is intended for use with the RaspPi. I may have this camera from a different project.
In any case, I was able to calibrate the ov5647, on a Raspberry Pi 4B, running Debian Bullseye, docker server 27.0.3, and ROS Humble. I then ran apriltag_ros and now see much closer values for transform.translation.z. The translation.z value published by apriltag_ros is consistently 22% less than the actual distance from the camera to the tag.
I have been using an 8x10 calibration checkerboard with 15 mm squares. I suspect that might be a bit small, since the calibration process didn't detect it at more than 40 cm from the camera. I'll print a much larger checkerboard, re-calibrate, and see if that improves the translation.z accuracy.

christianrauch, 2025-03-13:
> I'm not certain the ov5647 is intended for use with the RaspPi.

Well, the documentation clearly shows that this sensor is used by the "Camera Module v1". This is a fixed-focus camera module. If you can use the camera module with libcamera, you should be fine.

> The translation.z value published by apriltag_ros is consistently 22% less than the actual distance from the camera to the tag.

Could it be that either the checkerboard or the tag dimensions were measured incorrectly? See https://github.com/AprilRobotics/apriltag?tab=readme-ov-file#pose-estimation for the correct way to measure the tag edge length.

> I have been using an 8x10 calibration checkerboard with 15 mm squares. I suspect that might be a bit small, since the calibration process didn't detect it at more than 40 cm from the camera. I'll print a much larger checkerboard, re-calibrate, and see if that improves the translation.z accuracy.

Yes. The larger the calibration pattern, the better the accuracy. Also make sure that this is flat. Alternatively to the [`ros2 run camera_calibration cameracalibrator`](https://docs.ros.org/en/rolling/p/camera_calibration/doc/tutorial_mono.html), which is just using OpenCV, you can also try out [Kalibr](https://github.com/ethz-asl/kalibr/wiki).

billmania, 2025-03-13:
Doh!
I continually harangue anyone within hearing about the wonders of the metric system, since it's only necessary to shift the decimal point. Then I go and set the square size parameters for my 15 mm squares to 0.15!  (-_-)
I've made a new calibration checkerboard with 0.033 m squares and will re-calibrate all my cameras now.

billmania, 2025-03-17:
I found a Logitech QuickCam Pro 9000 USB 2.0 camera. I configured it for YUYV and 1600x1200 resolution. I calibrated it using the camera_calibration package and then tested it with a 15.3 cm version of tag 36h11 ID 586. The results were:

ActualDistance  ReportedDistance PercentError
72 cm                               85 cm                   18%
106 cm                            127 cm                  10%
169 cm                            196 cm                  16%

For my Use Case this will be "good enough". I'll just multiply by a constant to correct the translation.z value.

christianrauch, 2025-03-28:
I assume your issue is related to calibration. I am going to close this now.

closed 2025-03-28 by christianrauch

billmania, 2025-03-28:
Yes, I now believe it was a combination of using a camera with high distortion from a fisheye lens AND a calibration error on my part. I'd still like to get better accuracy on the Z axis, but it's good enough for now. Thanks for your help.