christianrauch/apriltag_ros #49: Build Failed (ROS2 foxy)

Harin99, 2025-05-07:
Dear maintainer @christianrauch, 

First of all, thank you for this great repo. 

I am trying to run apriltag_ros in an Ubuntu 20.04 environment with ROS 2 Foxy. However, I encounter the following error during the rosdep installation process.

### Steps for reproduce 
1. Perform like below: 

```jsx
mkdir -p ./apriltag_ros_ws/src 
cd apriltag_ros_ws/src 
git clone https://github.com/christianrauch/apriltag_ros.git
git clone https://github.com/AprilRobotics/apriltag.git
git clone https://github.com/christianrauch/apriltag_msgs.git 
git clone https://github.com/christianrauch/camera_ros.git
git clone https://git.libcamera.org/libcamera/libcamera.git
rosdep install --from-paths src --ignore-src -r -y
```

2. Error happens like below: 

```jsx
ac@AC:~/apriltag_ros_ws$ rosdep install --from-paths src --ignore-src -r -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
camera_ros: Cannot locate rosdep definition for [libcamera]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

```

3. To resolve this, I referred to (https://github.com/christianrauch/camera_ros) and created a separate workspace to follow the provided steps. However, I was not able to build the package successfully. 

```bash
mkdir -p ~/camera_ws/src
cd ~/camera_ws/src
pip3 install colcon-meson # I got the error when perfoming (sudo apt -y install python3-colcon-meson)
git clone https://github.com/christianrauch/camera_ros.git
cd ~/camera_ws/
rosdep install -y --from-paths src --ignore-src --rosdistro foxy --skip-keys=libcamera
colcon build --event-handlers=console_direct+
```

I got below error: 

```bash
[1.067s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'meson' in 'src/libcamera/utils/tuning/libtuning/modules/lux': __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 144, in _identify
    retval = extension.identify(_reused_descriptor_instance)
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 122, in identify
    parser = CustomInterpreter(desc.path, "", "")
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 26, in __init__
    super().__init__(source_root, subdir, subproject)
TypeError: __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'

[1.069s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'meson' in 'src/libcamera/utils/tuning/libtuning/parsers': __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 144, in _identify
    retval = extension.identify(_reused_descriptor_instance)
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 122, in identify
    parser = CustomInterpreter(desc.path, "", "")
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 26, in __init__
    super().__init__(source_root, subdir, subproject)
TypeError: __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'

[1.070s] ERROR:colcon.colcon_core.package_identification:Exception in package identification extension 'meson' in 'src/libcamera/utils/tuning/raspberrypi': __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/colcon_core/package_identification/__init__.py", line 144, in _identify
    retval = extension.identify(_reused_descriptor_instance)
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 122, in identify
    parser = CustomInterpreter(desc.path, "", "")
  File "/home/ac/.local/lib/python3.8/site-packages/colcon_meson/identification.py", line 26, in __init__
    super().__init__(source_root, subdir, subproject)
TypeError: __init__() missing 2 required positional arguments: 'subproject_dir' and 'env'

Starting >>> camera_ros
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.18 or higher is required.  You are running version 3.16.3

-- Configuring incomplete, errors occurred!
--- stderr: camera_ros
CMake Error at CMakeLists.txt:1 (cmake_minimum_required):
  CMake 3.18 or higher is required.  You are running version 3.16.3

---
Failed   <<< camera_ros [0.09s, exited with code 1]
                                
Summary: 0 packages finished [1.04s]
  1 package failed: camera_ros
  1 package had stderr output: camera_ros
```

### Questions 
From the final error message, it seems that I need to upgrade my CMake version. However, I would like to ask if there is any way to use `apriltag_ros` within the ROS 2 Foxy environment without upgrading CMake.

I would greatly appreciate any advice or suggestions you could provide.

christianrauch, 2025-05-07:
You can tell rosdep to skip the `libcamera` and `camera_ros` keys. See the rosdep documentation.

`libcamera` is not available on Ubuntu 20.04. But this repo isn't supporting EOL Ubuntu and ROS distros anyway.

Harin99, 2025-05-07:
@christianrauch Thank you very much for the prompt response.

I will try running rosdep after skipping camera_ros and libcamera. 

Additionally, I would like to use apriltag_ros in a ROS 2 Foxy environment.
Given that I cannot change my system setup, would it still be possible to run apriltag_ros without camera_ros and libcamera?

christianrauch, 2025-05-07:
> Additionally, I would like to use apriltag_ros in a ROS 2 Foxy environment.
> Given that I cannot change my system setup, would it still be possible to run apriltag_ros without camera_ros and libcamera?

Maybe. I don't know. There are no CI tests for Ubuntu 20.04. If something breaks, I wouldn't know.

Why can you not update your OS to a newer version?

I can only recommend moving away from unsupported distributions, as you will be on your own with fixing issues. Eventually, those ROS Debian packages will also disappear. If you cannot update your host OS, you can still run the Debian packages for newer Ubuntu versions in a Docker or other container system.