christianrauch/apriltag_ros #51: using realsense camera with apriltag_ros

RosaZheng, 2025-08-19:
Dear maintainer @christianrauch,
I am also trying to use your ROS2 wrapper on Humble with a Realsense camera D435. Would you provide  guidance on how to get the Realsense camera to work with apriltag detection? I modified the [launch.yml here](https://drive.google.com/file/d/1YWM0TpKrg6YLBJ6GL--WhhZ_wfL0ejna/view?usp=sharing) and the Realsense camera seems to work, but there is no detection results in /apriltag_ros/detections. How do I visualize the results using rviz2?

christianrauch, 2025-08-19:
Sorry, I cannot provide extensive general ROS support here in my free time. If you need help with generic ROS issues, have a look at https://robotics.stackexchange.com. This issue tracker is only meant for issues with the node itself. The package has all its dependencies defined and it follows the usual ROS procedures for resolving dependencies and compiling. You can also just install the binary packages on Ubuntu (e.g. `sudo apt install ros-jazzy-apriltag-ros`), to resolve the dependencies automatically.

RosaZheng, 2025-08-19:
OK. Thanks. Found the documentation: https://docs.ros.org/en/humble/p/apriltag/ will follow it.

closed 2025-08-19 by RosaZheng